zbigniewmotyka/home-assistant-solplanet #12: Additional device (=3), what could it be?

Lafferfr, 2024-10-16:
Hello,

I guess on device=3 is also something, I was not yet able to identify it. The answer is:

{"flg":1,"tim":"20241016122122","pac":-128,"itd":284,"otd":284,"iet":452,"oet":600,"mod":1}

Lafferfr, 2024-10-17:
Hi, as a improvement report: in the inverter values, there is apparent power, which has as UNIT -> VA, but VA is the same as W(att). Could you change the definition into W(att) instead of VA(Volt*Ampere)? Because if it has a different unit, it can not be used in statistics vs the other Values which are in W(att).

Thx

zbigniewmotyka, 2024-10-17:
Hi, unit VA for apparent power is correct. Apparent power is measured in the volt-ampere unit instead of watt.

Lafferfr, 2024-10-21:
VA equals W, the definition of 1 Watt is 1 Volt * 1 Ampere. Changing it, would help to use it in Statistics without introduction of a Helper.

andpaw88, 2024-11-01:
> Right, as I know, this device is energy meter connected to inverter (in most cases, it's ct clamps). It will also be supported by this integration.

When do you expect this? I have EASTRON SDM630-Modbus V2 connected to my inverter. If i can help somehow with data from it just let me know

zbigniewmotyka, 2024-11-02:
Meter device is next on my list, so if everything with battery support is ok (latest `v0.2.0-beta5` version is the last expected beta version before releasing `v0.2.0`) then meter device support will be added in max 2 weeks.

zbigniewmotyka, 2024-11-10:
Meter support is prepared in `v0.3.0-beta1` version. Please check if it works with your inverter.

zubir2k, 2024-11-11:
> Meter support is prepared in `v0.3.0-beta2` version. Please check if it works with your inverter.

Meter works. The detail of the meter is missing though
http://`ip_address`:8484/getdev.cgi?device=3

- manufactory
- name
- sn

```JSON
{"mod":2,"enb":1,"exp_m":0,"regulate":5,"enb_PF":0,"target_PF":0,"total_pac":1056,"total_fac":6000,"meter_pac":18,"sn":"EASTRONB3240Dxxxxxx","manufactory":"EASTRON","type":"1","name":"SDM230","model":2,"abs":0,"offset":0}
```

zbigniewmotyka, 2024-11-11:
Thanks for sharing this response. My inverter doesn't send these values. I will add support for them.

andpaw88, 2024-11-11:
It works for me:
[image]

Lafferfr, 2024-11-11:
It works for me too.... Great work!

zbigniewmotyka, 2024-11-11:
In version `v0.3.0-beta3` support for `manufactory`, `name` and `sn` is provided.

zubir2k, 2024-11-11:
Cool! it's now showing the meter's info 👍

> In version `v0.3.0-beta3` support for `manufactory`, `name` and `sn` is provided.

zubir2k, 2024-11-13:
I found that the value is slightly off. The unit as shown in Solplanet portal is in MWh while the sensor is in kWh.
The value in kWh should be `1555kWh` which is equivalent to `1.555MWh`

`{"flg":1,"tim":"","pac":-381,"itd":0,"otd":0,"iet":15546,"oet":13413,"mod":2,"enb":1}`

since the api value is `15546` I guess it should be divided only by 10 making it `1554.6`

[image]

[image]

zbigniewmotyka, 2024-11-14:
Version `v0.3.0-beta4` contains @zubir2k fix.

zbigniewmotyka, 2024-11-15:
Meter support is now released as `v0.3.0` version

closed 2024-11-15 by zbigniewmotyka