zbigniewmotyka/home-assistant-solplanet #14: Integration needs to be reloaded after inverter goes offline

krisdieteren, 2024-10-17:
My inverter (ASW6K-LT-G2 Pro) goes offline each evening. If I don’t reload the integration in the morning after it is online again, the sensors remain unavailable. 

Is it possible to reload automatically?

blasteris, 2024-10-17:
I have the same problem with my (ASW25K-LT-G3) inverter, thou I don't need to reload every morning.

krisdieteren, 2024-10-17:
Sometimes it runs for a few days, but I have not found a logic in it yet.

tomz71, 2024-10-19:
Same here with ASW6K-LT-G2 Pro inverter. It works for few days without issue, but sometimes it needs to reload.

blasteris, 2024-10-19:
I think this happens when solar panels stop producing energy and inverter goes to sleep and while it is in sleep mode you reload or restart HA. After restart it stops detecting sensors.  I created automation to reload config entry 25min after sunrise.

tomz71, 2024-10-19:
> I think this happens when solar panels stop producing energy and inverter goes to sleep and while it is in sleep mode you reload or restart HA. After restart it stops detecting sensors. I created automation to reload config entry 25min after sunrise.

Yes, it makes sense, worked whole week without problems, yesterday night i have configured and restarted the HA, and today morning it was offline. Thanks for the light-up! :)

zbigniewmotyka, 2024-10-20:
Can you attach logs for Solplanet integration?

krisdieteren, 2024-10-21:
Also from me a big thanks for this integration! I will monitor if not restarting HA while the inverter is offline solves the issue.

krisdieteren, 2024-10-25:
Seems that indeed running a HA update while the inverter is offline causes the issue.

commit 6388dfb4 (2025-01-06): Fix #14 and exceptions during inverter sleep when smart meter not installed (#29)

* Fix issue #14

* Fix exceptons and sensors unavailable during night if meter is not installed

* add missing logger definition

* update entity value only if there is data

* Update custom_components/solplanet/coordinator.py

---------
closed 2024-10-25 by krisdieteren