zbigniewmotyka/home-assistant-solplanet #15: Sensor Names should be preceaded by Solplanet

Lafferfr, 2024-10-17:
Hi, it would be nice if the Sensor Names would be instead of e.g.

Apparent Power -> Solplanet Apparent Power
MPPT 1 power -> Solplanet MPPT 1 Power

zbigniewmotyka, 2024-10-17:
As I understand, the main reason for this change is to distinct source of sensors data when we need to select entity (f.ex. on energy dashboard) and we have more than 1 device with the same sensor name. Sensor name could be prefixed with "Solplanet" but it doesn't resolve the problem when we add 2 inverters to home assistant. This case can be resolved with adding new field "Sensor name prefix" on configuration screen, but there is another case which won't be addressed using this method - under single IP address there could be more than one inverter, so every device will get the same prefix. 

There is no method (or maybe I don't know it) how to solve it in complex way.

Luckily, Home Assistant has a built-in feature which allows changing the name of the entity.

Lafferfr, 2024-10-21:
I agree that this is a solution, but 2 inverters are less common, than having only 1. But I understand your argument, it was just a convenience feature.

closed 2024-10-23 by zbigniewmotyka