zbigniewmotyka/home-assistant-solplanet #17: Error on grid-tie system without battery

203agent203, 2024-10-21:
Hi there,
Installation failed with the error "failed setup, will retry. Error fetching data from API:'battery'".
I'm using the latest beta.
I don't have battery in my system.
Please let me know if you need any further logs.
Cheers,
T.

zbigniewmotyka, 2024-10-21:
Which version have you installed? If you have 0.2.0-betaX, then it won't work for you. This version right now is used only for inverters supporting battery. Because of that, those releases are marked as pre-release to not be installed during integration update. The next beta release will contain some mechanisms to check battery support.

zbigniewmotyka, 2024-10-21:
Use 0.1.x version for your setup

Lafferfr, 2024-10-21:
How should I update to the newest Beta (I have Battery and Powermeter)?

closed 2024-10-22 by 203agent203

zbigniewmotyka, 2024-10-22:
> How should I update to the newest Beta (I have Battery and Powermeter)?

In hacs when you select integration installation you need to manually select version.

[image]