zbigniewmotyka/home-assistant-solplanet #18: Adding support for EV-chargers ?

oscarhemberg, 2024-11-10:
Hi,
I have a large Solplanet installation and just found your integration. Installed perfectly and found both my inverters and both battery stacks and everything seems to run well. Just wanted to say thank you.

If you ever decide to add support also for their EV-chargers let me know. I have 2 of those as well and would be happy to help test.

Keep up the good work.

zubir2k, 2024-11-11:
I am also curious which code represent EV charger. Perhaps could you share the output from this api:
http://`inverter ip`:8484/getdev.cgi?device=`code`

Code that we know so far:
1 = wifi dongle
2 = inverter
3 = smart meter
4 = battery

you could try with `0` or `5` and see if there's any response

oscarhemberg, 2024-11-11:
I get the same response with 0 and 5,6,7,... but they are just the dongle
and same response. I however get no response with 1.

The EV chargers have separate IPs and connected to cloud directly I believe
and not through dongle. I tried using the call with the EV charger IP but
it refuses to connect.

zbigniewmotyka, 2024-11-11:
@oscarhemberg, how you can configure this ev charger (f.ex. to specify the WiFi network)? During this configuration, the solplanet application connects to the charger using WiFi? If yes, then it must be possible to connect somehow to this charger. If it is configured using another method (f.ex. Bluetooth ble) then there is a risk that it could be hard to read data in a similar way as for inverter.

oscarhemberg, 2024-11-11:
They support BLE, wifi or Ethernet (and I think also RS585 but that I have
not tried). I have mine with ethernet cable and they have separate IPs just
through normal DHCP. Setup in the Solplanet App by scanning the QR code
same as for the inverters and batteries. So for sure possible to connect to
them but I just dont know what port and command they listen to.

--Oscar

zubir2k, 2024-11-11:
@oscarhemberg if the charger has indeed linked to the Solplanet app, then the local API should return with the EV charger values. Perhaps you could try login to the web (https://internation-cloud.solplanet.net/user/login) and see if you're able to view the EV charger there.. 

[image]

oscarhemberg, 2024-11-11:
Yes, I see them there.

//Oscar

zbigniewmotyka, 2024-11-11:
A good point to start would be to use nmap (https://nmap.org/) to check if the charger has any open ports by running command `nmap -vv -p- <charger-ip-address>` (f.ex. `nmap -vv -p- 192.168.1.100`).

zubir2k, 2024-11-12:
> I get the same response with 0 and 5,6,7,... but they are just the dongle and same response. I however get no response with 1. The EV chargers have separate IPs and connected to cloud directly I believe and not through dongle. I tried using the call with the EV charger IP but it refuses to connect.

@oscarhemberg Come to think of it, the response that you got for device code = 0, 5-7 is pointing to the wifi dongle instead of code 1. In my case, I got 404 response if I tried calling device code 0, 5, 6 and 7. Perhaps maybe if you could share the json that you got earlier for these codes.  Logically, if connected OK in Solplanet, the data should be available.

@zbigniewmotyka imho, its best that we stick to 1 source of connection. Unless of course the Solplanet logger doesn't include EV charger to begin with 😉

oscarhemberg, 2024-11-12:
Attached are the JSON files. for code=1 I just get "request not found, 404"
reply. and for 6,7,8... It is the same as 5. WHen I look at them I get
confused as I can only see the response of one battery stack. I have two
inverters in parallel and each has a separate battery stack. In the App/web
everything shows up correctly including the smart meter and 2x battery
packs. Also in HA the batteries both show up although I can not see one of
them in the JSON files. I also tried the Nmap but could find no open
ports on one of the EVchargers.

--Oscar

[image: Screenshot 2024-11-12 074546.png]

zubir2k, 2024-11-14:
> Attached are the JSON files. for code=1 I just get "request not found, 404" reply. and for 6,7,8... It is the same as 5. WHen I look at them I get confused as I can only see the response of one [...]

Hi Oscar @oscarhemberg, your response earlier doesnt have the JSON file. could you resend it again here. Thanks

Tjepp, 2024-11-16:
Hi! I also have a Solplanet charger and would love this integration. Any way I can help? 😄

zubir2k, 2024-11-16:
Hi @Tjepp 
Could you help to check what response did you get when accessing to this API?
Try with code `0` or `5` and see if there's any response.

The API: http://`inverter ip`:8484/getdev.cgi?device=`code`

Code that we know so far: 
1 = wifi dongle 
2 = inverter 
3 = smart meter 
4 = battery

[image]

Tjepp, 2024-11-17:
Both 0 and 5 replies "request not found, 404"

M3nt0san, 2025-03-08:
Hello, 
Did anyoane manage to use Solplanet chargers with the integration? I can also help with some testing if there's someone willing to help out :).
Except for this fact, the integration works fasntastic for me, Kudos @zbigniewmotyka.

zbigniewmotyka, 2025-03-09:
@M3nt0san Thanks! Glad to hear that — appreciate the feedback!

Right now, there is no known way to get data from the EV charger. If you find a way to read these values, some instructions would allow me to try adding support.