zbigniewmotyka/home-assistant-solplanet #27: User input

zamzon, 2024-12-08:
First of all, thanks for this integration,

Are there any plans to make it possible to control the inverter from HA
Like start/stop charging the batteri, change user mode ect.?

How do you communicate with the inverter, is it locally or is it from the cloud ?

zbigniewmotyka, 2024-12-08:
Hi, there are plans to make possible to control the inverter. Right now there is a pull request with support for changing battery mode: https://github.com/zbigniewmotyka/home-assistant-solplanet/pull/25 - it's currently in beta version so if you want to test it you have to manually select `v0.4.0-beta1` version during integration installation.

This component uses local communication with inverter. I don't want to rely on any cloud service to control devices in my house.

zamzon, 2024-12-08:
Hi

I have now tried it, and it works perfectly with my inverter 10kw hybrid T2
And I have tried changing state in the Solplanet app, and the state changes in HA. So all looked perfect.

I have connected my inverter to modbus and with ESPhome. I can change all states on the inverter.
So it would be nice if I could use your integration instead.

So what I need of your integration  to make it perfect is, to control the charge and discharge power, and discharge SOC lower limit. This it what I use to charge the battery and for keeping the power for later use. I set the SOC lower limit to 99%, and then the battery won't discharge until I set it to 10% in self-consumption mode

I did attached a image with the modbus register, if it could be of  any help 

I am happy to help you with some testing ?
How do you send this commands, is it some Get/put commands  ?

[image]

zbigniewmotyka, 2025-01-15:
Hi @zamzon 

I've just released new beta version `v0.5.0-beta1` which adds feature to control soc min and max. Let me know if it works correctly for you.

zamzon, 2025-01-15:
Hi

Thanks for reaching out, I have testet it, and it seems to work as it
should. if I set a value in the app, it changes automatically in HA and
vice versa
Is it possible to set the SOC min higher than 45%

I control the inverter trough Modbus, and here it is possible to set a
value of 99%
Because I didn’t find a good way to set the system in standby mode after
the battery is charged up. And by setting it to 99% the system will not
discharge billow

How about the Charge/discharge power, are you also thinking of including
this.

ons. 15. jan. 2025 kl. 21.43 skrev Zbigniew Motyka ***@***.***

ffavero, 2025-03-07:
Hi @zbigniewmotyka,

Thank you for this great integration! There is no doubt that this is the best way to interact with our Solplanet system, even compared to the official mobile app (though I know the bar isn't very high to begin with).

I live in Denmark, where we have variable electricity prices each hour. I have sensors for both selling and buying prices, which is crucial for managing the system with the SOC levels. During certain peak hours, especially in the morning when the sun is shining and no one is home, I'd like to be able to dump all the battery leftovers into the grid and recharge it later when the selling price is close to zero.

Currently, I find it impossible to force the battery to discharge using the integration. The only way I can do this is by setting up a discharge schedule in the app and then switching to custom mode.

Would it be possible to implement charge and discharge commands in the integration?

Thanks again for your hard work!

ffavero, 2025-03-07:
To provide more context in addition to what @zamzon posted, here is the [full PDF on the Modbus API](https://github.com/user-attachments/files/17982869/MB001_ASW.GEN-Modbus-en_V2.1.4.pdf), uploaded by another user.

Playing with random requests can be quite dangerous. A good starting point would be to take something we are already doing with the current implementation, such as any of the read-only operations, and use the fdbg.cgi endpoint to achieve the same results.

This [great idea to use a Node-RED proxy to/from Home Assistant](https://community.home-assistant.io/t/api-integration-for-solplanet-inverter-aiswei/569754/43) and the inverter looks very powerful. However, it might be overkill since we already have a pretty good monitoring system in place. We just need to send a few additional instructions to the inverter.

ffavero, 2025-03-08:
I ended up setting up NodeRed to test the modbus approach, indeed it works, but it makes the communication with the inverter very choppy, all http requests are slow/stuck, so the integration and the modbus also get stuck, both. I suspect either I'm making too many requests, with both integration setup, or in general, the fdbg.cgi endpoint is very slow/expensive.

I even tried with the solplanet integration disabled, and I couldn't make it work fully.

I've managed to see few things though, my modbus configuration looked like this
```

- name: modbus_solplanet
  type: rtuovertcp
  host: 127.0.0.1
  port: 502
  sensors:
    - name: Active Power
      unit_of_measurement: W
      slave: 3
      address: 1370
      input_type: input
      data_type: int32
      min_value: 0
      scan_interval: 30
    - name: battery_status
      slave: 3
      address: 1607
      input_type: input
      data_type: int16
      scan_interval: 90
    - name: charge_flag
      slave: 4
      address: 1151
      input_type: holding
      data_type: int16
      scan_interval: 60
    - name: charge_cmd
      unit_of_measurement: W
      slave: 4
      address: 1152
      input_type: holding
      data_type: int16
      scan_interval: 60
```

zamzon, 2025-03-08:
Hi

I have a working setup with modbusRTU and this Solplanet integration.

I use a ESP32 and ESPHome it works perfectly

ffavero, 2025-03-09:
> Hi
> 
> I have a working setup with modbusRTU and this Solplanet integration.
> 
> I use a ESP32 and ESPHome it works perfectly

Nice! what do you mean with "this Solplanet integration", is modbus running in parallel for few extra sensors/inputs, or have you converted the whole integration to work via modbus?

Are you able to set in charge/discharge mode? if so, could you share the relevant modbus configuration?
Thanks a lot!

ffavero, 2025-03-09:
there is also the `/getdefine.cgi` endpoint.

however I don't think I fully understand the data. I get

```json
{"Pin":4000,"Pout":3100,"Sun":[0,0,0,0,0,0],"Mon":[0,0,0,0,0,0],"Tus":[0,0,0,0,0,0],"Wen":[0,0,0,0,0,0],"Thu":[0,0,0,0,0,0],"Fri":[0,0,0,0,0,0],"Sat":[302005251,0,0,0,0,0]}
```

I have it set it to charge at 4000w and discharge at 3100w. And as schedule there is Saturday (and at this point I understand the data), 1 hour from 18:00 to 19:00. Trying different decoding strategies I get something like this to convert it to a human readable format:

```python
def decode_inverter_schedule(code):
    """
    Decode inverter schedule code to extract:
    - Status (charge/discharge)
    - Start hour
    - Duration (in hours)
    
    Returns a tuple of (status, start_hour, duration)
    """
    # Convert to binary and ensure 32 bits
    binary = bin(code)[2:].zfill(32)
    
    # Extract start hour from first byte
    start_hour = int(binary[:8], 2)
    
    # Extract status from last bit
    status_bit = int(binary[-1])
    status = "discharge" if status_bit == 1 else "charge"    
    
    # The second set of 8 bits (bits 9-16) seems promising
    duration_bits = binary[8:16]
    
    
    # Let's try the third set (bits 17-24)
    duration_related = binary[16:24]
    
    # Let's derive an algorithm:
    duration_value = int(duration_related, 2)

    
    duration = duration_value // 60
    
    return status, start_hour, duration

```

If the `setting.cgi` also offers a way to change the schedule data, I can see it could be possible to have a somewhat convolute control even without exposing directly the charge/discard

zbigniewmotyka, 2025-03-09:
Hi @ffavero,

Thanks for sending the full Modbus API. On a separate branch, I have a working solution to request data from Modbus using `fdbg.cgi`. A request to this script takes between 0.3s and 1.5s, so it shouldn't be used to query data for many sensors. Rather, `fdbg.cgi` should be used to set some data. 

[image]

However, I have a problem — I’m able to set up a middleware between the Solplanet app and the inverter to capture the Modbus frame actually sent from the app to the device, but the app doesn’t allow setting data in registry 41153 (which controls charge or discharge power). 

Would it be possible for you to send me an example RTU frame in hex format (like in the screenshot above) for both charge and discharge power?

zbigniewmotyka, 2025-03-09:
@ffavero I’ve checked the `/getdefine.cgi` and `/settings.cgi` endpoints, and it looks like they could be used to read or change the schedule. However, this approach would be more complex than sending a Modbus frame to set charge or discharge power, so IMHO it should be treated as plan B.

zamzon, 2025-03-09:
Hi
Yes I am able to set the charge and discharge watt and work mode with modbusRTU

[image][image]

ffavero, 2025-03-10:
I've started with a bunch of encoded messages from the debug of the Node-RED approach, and quickly ended up creating a whole debugging script. With this, I was able to set the battery to discharge!

https://gist.github.com/ffavero/bcca06951b5023d3d3dbb6a9f25152c0

I also realized that I was setting slave:4 for many of the values, which was causing most of the issues I was facing. without the cumbersome debug in home assistant + node-red, I quickly realized the correct slave is apparently always 3.

However, to use the charge/discharge functionality, the inverter needs to be set to custom mode. I struggled with this because the mode set via the app/HTTP endpoints is not the same as the one set by fdbg.cgi/Modbus, at least not entirely. Essentially, custom mode in the UI doesn't mean the same custom mode you can set via the Modbus API.

To set the battery into custom mode, use the following command:

```
python mb_rtu.py --host 192.168.86.37 --port 8484 --slave 3 --data-type int16 --address 1103 --type holding --query --write --value 4
```

Then, to start discharging roughly 1000W:

```
python3 mb_rtu.py --host 192.168.86.37 --port 8484 --slave 3 --data-type int16 --address 1152 --type holding --query --write --value 1000
```

Or to charge 1000W:

```
python3 mb_rtu.py --host 192.168.86.37 --port 8484 --slave 3 --data-type int16 --address 1152 --type holding --query --write --value -1000
```

To stop and revert to "default" mode:

```
python mb_rtu.py --host 192.168.86.37 --port 8484 --slave 3 --data-type int16 --address 1103 --type holding --query --write --value 2
```
I'm not sure if this works perfectly, as there are quite a few quirks to sort out, but this definitely opens up a lot of options.

ffavero, 2025-03-11:
Just because it was mindbuggling, I got a stab at decoding how those values in the schedule could be turn into hours, duration and charge/discharge.
I made this script that was all around pretty accurate in setup the schedule, however it can only setup 1 out of 6 values per day (my use case doesn't use more that 1 op a day).
[here is the code,](https://gist.github.com/ffavero/dffd628d55e9765180e945682941d89f) it would be nice to integrate the scheduling anyway in the integration

ffavero, 2025-03-11:
Thanks @zbigniewmotyka !
I'll definitely try ASAP. I can see like this it gives a lot of flexibility to create automation setting the specific modbus registers, but I feel it's a bit dangerous to leave the service to be able to call any register, but again, I don't think this is a real issue.

What do you think in implementing the scheduling? this also can be set by a service to clear/set the schedule. I'm trying to prepare something for a pull request but I might be far from any working solution.

Thanks a lot for this! (although the sun seems to have taken a break this week, so not much to test :P)

ffavero, 2025-03-12:
@zbigniewmotyka  I've issued a PR with the schedule service, and I've now merged it with the modbus branch, so I can test both.
@zamzon, I guess you are from Denmark? Are you using any automation to decide the charge/discharge patterns based on weather (or forecast production) and spot price?

Thanks again

zamzon, 2025-03-12:
Hi

Yes I am also from Denmark.

Sunmate.io is working on integrating Solplanet, there should be a solution in this month.
There is a sunmate integration to home assistant, so this could be a way to automate your Solplanet inverter 

https://sunmate.io/?fbclid=IwZXh0bgNhZW0CMTEAAR3EHxtIWEVjvTfeZ21ReFYt_Nhl1WW3pbXC-wJljrid9IN3Tip0Ghz7Etg_aem_HYKIg6kvLSegmuhvPbQQaA

zbigniewmotyka, 2025-03-12:
@ffavero Thanks for your contribution! I’ve checked it, and it works almost perfectly. I’ve added a comment to your PR to address two issues — once those are resolved, I’ll prepare a beta release.

zbigniewmotyka, 2025-03-13:
@ffavero I've merged your PR. New version `v0.7.0` is released but you have to select that version manually during installation. After you confirm that it works ok (also with modbus) I will mark this version as latest one.

zbigniewmotyka, 2025-03-13:
I marked this version as latest one so you can just update this integration.

ffavero, 2025-03-13:
That's great!
I've tested the schedule features this morning, and they worked perfectly! I think someone could even create a nice dashboard card with the schedule attributes.
This integration is now doing more than what the mobile app does. :)

I changed the Pout slider while the scheduled discharge was running, and the inverter immediately increase the power accordingly, so it's really nice and more useable than the app

I am currently testing the Modbus service. The "full address, not offset" means we don't have to subtract 1 from the register. For example, if the holding register in the documentation is 41153, this translates to 1152. So, do we need to set it to 1153 instead? to be safe I think I'll run it in dry mode for now.

Regarding the Modbus service, I should have done the same in the schedule service. The entity ID is not really required, but I felt it needed to be for some reason. The schedule is on the inverter, and there is no way to specify which battery (if multiple batteries are even supported) to charge/discharge. So, the target is really unnecessary.
An additionaly inverter would means a new ip with its own integration

ffavero, 2025-03-13:
@zbigniewmotyka I can confirm that the Modbus service works as intended. The addresses to use are the plain values from the AISWEI Modbus API documentation. It's important to ensure that the correct codes and values are used to avoid any potential unwanted side effects. As with any powerful tool, responsible use is key.

This opens up full control of the inverter. I wonder if there are contingency measures in case the wrong default value gets altered. In theory, it would be possible to save all the responses for every value to get a default/current table states.

zbigniewmotyka, 2025-03-13:
@ffavero I realized that my implementation of service registration is incorrect, so please don’t change your implementation to match mine! 🙂 I need to refactor the Modbus service to include a field for the entity ID. Let me explain why:

1. **Service registration per entry:**
The service should be created only once during system setup for all configured devices. Currently, I’ve placed the async_register service call in async_setup_entry, which means the service gets registered for every entry. I haven’t tested it with multiple entries yet, but I suspect the service might only appear once due to the same name. If that’s the case, service requests might end up targeting only the first or last configured entry.

2. **Entity ID vs Modbus address:**
Providing a Modbus address is more complex than selecting an entity ID. When accessing the `http://<ip>:8484/getdev.cgi?device=2` endpoint, you’ll see a property called `add` containing the Modbus address. This address could be automatically retrieved, but it needs to be linked to the parameters passed to the service.

3. **Multiple inverters on one Wi-Fi dongle:**
In scenarios where multiple inverters share a single Wi-Fi dongle, each inverter has its own Modbus address. In this case, selecting the entity ID is far simpler than manually entering the address in the service call.

I’m also thinking about how to protect users from accidentally causing unwanted side effects when modifying Modbus registers. One idea is to hardcode a list of "safe" registers within the integration. If someone wants to change a register outside of that list, they’d need to provide the address in the entry configuration form — along with a mandatory checkbox like "I understand that this action might damage my device".

Let me know what you think! 🚀

ffavero, 2025-03-13:
@zbigniewmotyka I can see how managing multiple device installations could become very complex

The warning notice toggle is a great idea for the Modbus service. However, enforcing and documenting a list of safe addresses would add complexity on using the service.
Instead of expecting users to manage this, we could define a few Modbus-based services for key functionalities. For any additional tweaks, users could enable an "I take responsibility" toggle and use the modbus service directly.

As a novice with the inverter, the only missing feature I notice in the main integration is start/stop charging. I wouldn’t want to mess with overvoltage or frequency settings, though more experienced users might.

To make it more intuitive, a start charging service should take a positive value for charging and a negative value for discharging—opposite to how 41153 currently behaves. Also, "charging service" might not be the best name since it also allows discharging.
The service would:

1. Set 41153 to the specified value multiplied by -1.
2. Retrieve and store the current state of 41104.
3. Change 41104 to 4.

A stop charging service would then restore the previous state of 41104. This could be tied to a binary sensor or a toggleable number for easier control.
Bottom line, users don’t have to manually find the correct Modbus addresses for this at all.

jjbolhuis, 2025-07-21:
@zbigniewmotyka (and all other contributers): this integration is amazing!!

I'm from the Netherlands and here we have dynamic pricing on electricity. It would be nice to be able to turn the inverter off (or throttle it back) on specific times and days. I followed some instructions online and build myself a ESP32 board following this guide:

[https://www.vegt.dev/blog/link-your-solar-inverter-to-home-assistant-using-modbus-serial/](https://www.vegt.dev/blog/link-your-solar-inverter-to-home-assistant-using-modbus-serial/)

I'm not a coder, but with some help from ChapGPT I managed to create some code that creates in Home Assistant two entities (an 'on/off'-switch and a slider for throttling the inverter between 0 and 110%). It runs now om my home build ESP-board, but if I understand correctly, you can integrate it in this Solplanet integration (just like the battery status). 

Can you integrate this two functions (turn on/off and throttle (0-110%) as well? That would be awsome!

```
esphome:
  name: esp-modbus-inverter
  friendly_name: ESP-Modbus-Inverter

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

api:
  encryption:
    key: “PUT SOMETHING HERE”

ota:
  - platform: esphome
    password: “PUT A PASSWORD HERE"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Modbus-Inverter"
    password: “PUT A PASSWORD HERE"

captive_portal:

# Specifieke code voor module

uart:
  tx_pin: 8 # witte draad is tx
  rx_pin: 9 # GPI09 gele draad is rx

  baud_rate: 9600 # conform specs
  stop_bits: 1 # conform specs
  data_bits: 8 # conform specs
  parity: NONE # conform specs

modbus:
  id: modbus_PV

modbus_controller:
- id: modbus_controller_PV
  modbus_id: modbus_PV
  address: 3
  update_interval: 20s

switch:
  - platform: modbus_controller
    modbus_controller_id: modbus_controller_PV
    name: "PV Inverter ON/OFF"
    register_type: holding
    address: 200 #40201
    icon: "mdi:toggle-switch"

number:
  - platform: modbus_controller
    modbus_controller_id: modbus_controller_PV
    name: "PV Export limit RAW"
    register_type: holding
    address: 5402 #45403
    use_write_multiple: true 
    value_type: U_WORD
    min_value: 0
    max_value: 11000
    step: 100
    #unit_of_measurement: "%"
    #device_class: power_factor
```