zbigniewmotyka/home-assistant-solplanet #30: Control of IR Heater

rennradgoe, 2024-12-23:
Hi Zbigniew, 

Thanks a lot for the awesome HA integration. May I ask you for some support / explanation of displayed figures so I can get my automation-idea better realised. 

My idea is to run a infrared heater only while the PV panels generate enough power. My inverter is an ASW08kH-T3 with three strings and I use two of them actually. My thought is to add the values of _MPPT 1 power_  (sensor.solplanet_MySN_mppt_1_power) and _MPPT 2 power_ (sensor.solplanet_MySN_mppt_2_power) to start the IR heater from a certain level of PV power. 

However, I struggle to understand if the sum of _MPPT 1 & 2 power_ is the right value to make sure the the heater runs only with PV generated power. Within the entity list I have recognised also the value _PV Power_ (sensor.solplanet_battery_MySN_pv_power). This value is lower then the sum of _MPPT 1 & 2 power_. 

Can explain the difference between _PV Power_ and the sum of _MPPT 1 & 2 power_? Maybe you have also a good advice wich entity/ value can be used best for my idea. 

Thank a lot upfront and Merry Christmas 

Mat

zubir2k, 2024-12-27:
Hi

IMHO, the PV power sensor sensor.solplanet_battery_mySN_pv_power refers to the total generated power from the panels. One logical explanation as to why when you sum both mppt is higher compared to this is due to minor losses during DC to AC conversion. MPPT rightfully should be in DC.

Since you use generated power as trigger to turn on IR heater (which I assume you set to trigger after certain watts is generated), then I would suggest to use sensor.solplanet_mySN_power as it will show the power available for consumption after it's being deducted for battery recharging. You will see this value return to high (based on generated power at that time) the moment battery has fully recharged.

[image]

zbigniewmotyka, 2025-01-01:
Using `sensor.solplanet_mySN_power` as a trigger could be a good choice but not in this cases. The value of this sensor will increase during battery discharge, so requirements `run a infrared heater only while the PV panels generate enough power.` will not be fulfilled (as it also contains power from battery). There are 2 possibilities:

1. If infrared heater is more important than battery and home power consumption, then use `sensor.solplanet_battery_mySN_pv_power` sensor.
2. If you want to run an infrared heater only when you have some unused power, then you can use `sensor.solplanet_meter_mySN_grid_power` to check how much energy is sent back to the grid. F.ex. if your heater has 1kw power then you need to check if `sensor.solplanet_meter_mySN_grid_power <= -1000` (sensor has positive value when you take energy from grid and negative when you push it back to grid). Some tricky part here is to create a trigger to turn off the heater. When you turn on 1kw heater when `Grid power = -1000` then grid power will be `0` as no energy is pushed back to grid. To turn off, you need to use trigger `sensor.solplanet_meter_mySN_grid_power > grid_power_turn_on_value + treshold` where `threshold >= heater_power`

rennradgoe, 2025-01-05:
Hi Zbigniew, 

Many thanks for your feedback and sorry that I get back so late.
Your idea, to  monitor the grid power sound interesting. 

I played around a bit and think it might be also possible to combine the following both conditions as trigger. 

 1. _sensor.solplanet_battery_mySN_pv_power > 1000 W_ (e.g. for a 1000 W IR heater) 
    AND  
2. _sensor.solplanet_battery_mySN_battery_state_of_charge_ > XX (e.g. 90% SOC)
 
In this scenario the IR heater should only turn on when the PV panels generate enough power AND if the battery is charged to a certain level at the same time (90% in my example above). On the other side the IR heater will switch off, when one of the conditions are not met.

To make sure that not any grid power will consumed I can combine your idea as a third condition before the swith is triggered ON or OFF. 

What do you think about this option - your opinion is welcom. 

BR,
Mat

zbigniewmotyka, 2025-01-06:
@rennradgoe your idea with 2 conditions are ok if you are ok with situation when some power could be taken from grid.

Let's think about a situation where:
1. `sensor.solplanet_battery_mySN_pv_power = 1200W`
2. `sensor.solplanet_battery_mySN_battery_state_of_charge = 92%` (and you have configured trigger to run when soc >= 90)
3. `sensor.solplanet_battery_mySN_battery_power = -450W` (negative value means that battery is charged with that power)

In that case, you don't know how much energy you have free because you don't know:
1. inverter power consumption (my setup contains zamel mew-01 energy meter on AC line connected to inverter, and I've discovered that my inverter can consume even 350W of power - that's the difference between the value reported by inverter in `sensor.solplanet_lm010k09s2281273_power` sensor and my external energy meter)
2. home power usage

Adding third condition with `sensor.solplanet_meter_mySN_grid_power` sensor will not be enough. Let's add another assumption to example situation
1. `sensor.solplanet_meter_mySN_grid_power = -600W` (which means that you push 600W to the grid)

`grid_power` is not enough to power 1kW heater so condition based on this sensor prevents you from turning on heater, but if your battery work mode is set to `Self-consumption mode` then all power used to charge battery can be used to power the heater when it's turn on so you have `600W + 450W = 1050W` of unused energy.

In above situation we don't use pv_power as a real condition so it could be removed.
If you want to use battery SOC as a condition together with grid power then you need to make a more complicated rule than I proposed in previous comment with grid power.

closed 2025-01-26 by zbigniewmotyka