zbigniewmotyka/home-assistant-solplanet #36: Can not connect / install ?

JA16122000, 2025-01-25:
I can not connect my SolPlanet internal IP to Home Assistant:

[image]

[image]

[image]

[image]

[image]

What i'm doing wrong?

zbigniewmotyka, 2025-01-25:
Hi, I'm afraid that this inverter uses api in v1 version which is not supported by this integration :(
To confirm that please check which url below returns data (replace `ZS1500xxxxxxxxxxxx` with proper serial number):

1. http://192.168.178.82:8484/invdata.cgi?sn=ZS1500xxxxxxxxxxxx
2. http://192.168.178.82:8484/getdevdata.cgi?device=2&sn=ZS1500xxxxxxxxxxxx

JA16122000, 2025-01-26:
> Hi, I'm afraid that this inverter uses api in v1 version which is not supported by this integration :( To confirm that please check which url below returns data (replace `ZS1500xxxxxxxxxxxx` with [...]

None of above returns anything :(

JA16122000, 2025-01-26:
Now it is online, but both URL's don't do nothing

JA16122000, 2025-01-26:
can i somehow update the firmware, so they use then the newer API ?

zbigniewmotyka, 2025-01-26:
Updating software probably will not add support for the above urls. Can you check one additional url?
http://192.168.178.82/home.cgi

Did you try to use official zeversolar integration? https://www.home-assistant.io/integrations/zeversolar/

JA16122000, 2025-01-26:
> Updating software probably will not add support for the above urls. Can you check one additional url? http://192.168.178.82/home.cgi
> 
> Did you try to use official zeversolar integration? https://www.home-assistant.io/integrations/zeversolar/

http://192.168.178.82/home.cgi gives:

1 1 EAB9618C1643 XYGSSRKJJLCQTPHS M11 19703-826R+17829-719R 15:22 26/01/2025 0 1 ZS150060118C0137 43 1.57 OK Error 

Zeversolar integration just added a few hours ago, that also works indeed

zbigniewmotyka, 2025-01-26:
As I see, the response that you attached doesn't contain more data than official zeversolar integration provides, so I don't see a place to improve, and because of that this api won't be supported by my integration.
I hope everything will work fine with zeversolar integration.

closed 2025-01-26 by zbigniewmotyka