zbigniewmotyka/home-assistant-solplanet #37: Nie moge sie polaczyc z Inverterem

Andimuc79, 2025-01-28:
Mam Inwerter ASW3K-LT 
Probuje sie polaczyc ale caly czas otrzymuje blad:

[image]

Po poradach autora chcialem wlaczyc tryb debugowania, ale nie ma takiej opcji

[image]

zbigniewmotyka, 2025-01-29:
Zainstaluj proszę wersję `v0.5.0-debug` - jest w niej zmieniony sposób logowania zdarzeń dzięki czemu zapiszą się one w logach bez konieczności aktywowania trybu debugowania.
Po próbie dodania urządzenia przejdź do Ustawienia -> System -> Logi i kliknij w przycisk Pobierz Logi. Nie wiem ile logów zbiera Twój HA więc dla pewności zwiększyłbym liczbę linii do np. 1000 i dopiero wtedy pobrał plik. Jako że mogą w nim być wrażliwe informacje to prosiłbym o przesłanie go na maila: zbigniew.motyka@gmail.com. Zachęcam też do przejrzenia pliku i ewentualnego usunięcia linii z wrażliwymi danymi które nie są powiązanie z integracją solplanet.

Andimuc79, 2025-04-20:
Czesc
Zrobiles jakies poprawki?
Bo zrobilem update do tej najnowszej wersji, i dalej mam to samo
Cannot connected

Andimuc79, 2025-10-30:
[image]
Wpisuje IP falownika, i dalej to samo