zbigniewmotyka/home-assistant-solplanet #39: Works via Wi-Fi?

lexxnsk, 2025-02-10:
I found here https://community.home-assistant.io/t/api-integration-for-solplanet-inverter-aiswei/569754/58
the example of link:
resource: http://YOUR.IP.HERE:8484/getdevdata.cgi?device=2&sn=YOUR.SERIAL.NUMBER.OF.WIFI.DONGLE.OR.INVERTER**(ethern.only)**

Does it mean I can not use this API via Wi-Fi and should connect RJ45 to the dongle? How it should work?
I have ASW05KH-T2 inverter. connected via Wi-Fi.
And when I am trying to send curls am getting 404:

root@OpenWrt:~# curl http:// 192.168.1.113:8484/getdevdata.cgi?device=2&sn=BAxxxxxx
root@OpenWrt:~# request not found, 404

root@OpenWrt:~# curl http:// 192.168.1.113:8484/getdevdata.cgi?device=2&sn=OExxxxxx9
root@OpenWrt:~# request not found, 404

I’ve tried both SN (dongle and inverter).

lexxnsk, 2025-02-10:
looks like device4 gives info about inverter and battery together:

root@OpenWrt:~# curl http://192.168.1.113:8484/getdevdata.cgi?device=4 | jq
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   443  100   443    0     0   4769      0 --:--:-- --:--:-- --:--:--  4815
{
  "flg": 1,
  "tim": "20250210135656",
  "ppv": 2400,
  "etdpv": 177,
  "etopv": 634,
  "cst": 10,
  "bst": 1,
  "eb1": 65535,
  "wb1": 65535,
  "vb": 21330,
  "cb": 0,
  "pb": 0,
  "tb": 274,
  "soc": 99,
  "soh": 100,
  "cli": 150,
  "clo": 300,
  "ebi": 101,
  "ebo": 33,
  "eaci": 0,
  "eaco": 0,
  "vesp": 2348,
  "cesp": 3,
  "fesp": 4994,
  "pesp": 30,
  "rpesp": 0,
  "etdesp": 3,
  "etoesp": 23,
  "vl1esp": 2370,
  "il1esp": 3,
  "pac1esp": 18,
  "qac1esp": 0,
  "vl2esp": 2332,
  "il2esp": 3,
  "pac2esp": 16,
  "qac2esp": 0,
  "vl3esp": 2343,
  "il3esp": 3,
  "pac3esp": 0,
  "qac3esp": 0
}

zbigniewmotyka, 2025-02-13:
I don't know what `ethern.only` means, but as far as I know, an inverter connected by ethernet cable and by wifi works in the same way. In the sn parameter, you need to provide inverter serial number.

It's hard to say why you received not found error but if you are able to request data for device=4 then it should work also for device=2. Maybe you provide an invalid serial number? Try to check the following url: http://192.168.1.113:8484/getdev.cgi?device=2 - it returns a list of inverters connected to dongle with its serial number (`isn`).

Have you tried my integration to get data from inverter? Does it work for you?

closed 2025-03-09 by zbigniewmotyka