zbigniewmotyka/home-assistant-solplanet #44: Różnice w odczytach między HA a aplikacją Solplanet

Jurny123, 2025-03-15:
Cześć. Dodałem i skonfigurowałem dzisiaj Twoje rozszerzenie do mojego HA. Niestety to, co widzę w HA odbiega od tego, co pokazuje mi aplikacja - głównie w kontekście "energy distribution". Jest godzina 23:16, magazyn energii jest rozładowany do 10%, czyli zgodnie z konfiguracją - niżej nie zejdzie i cała energia pobierana jest z grida. Tak też mi pokazuje aplikacja Solplanet, ale z jakiegoś powodu w HA widzę, że 0.3 kWh jest pobierane z baterii. Czemu? Jak to skalibrować?

[image]

[image]

andpaw88, 2025-03-15:
Wg mnie 0,3 kWh to to co pobrane zostało z magazynu w ciągu dnia, a nie chwilowy pobór. Aktualizuje się to co godzinę.

Jurny123, 2025-03-15:
> Wg mnie 0,3 kWh to to co pobrane zostało z magazynu w ciągu dnia, a nie chwilowy pobór. Aktualizuje się to co godzinę.

W ciągu dnia magazyn był naładowany do pełna i wykorzystywana była energia z niego, aż do rozładowania do 10%. Sumarycznie więc powinno pokazywać około 8kWh w takiej sytuacji :)

andpaw88, 2025-03-15:
A kiedy uruchomiłeś integrację i podpiąłeś sensory pod kartę Energia? Od tego czasu mogłeś zużyć 0,3 kWh z magazynu?

Jurny123, 2025-03-15:
> A kiedy uruchomiłeś integrację i podpiąłeś sensory pod kartę Energia? Od tego czasu mogłeś zużyć 0,3 kWh z magazynu?

Hmm, to może być dobry trop. Integrację uruchomiłem około godziny 21:30, ale patrzę na wykresie, że w tym czasie magazyn miał 11% naładowania, więc mogło złapać te ostatnie "podrygi". Dobrze, zobaczymy w takim razie jutro, po pełnym dniu, co pokaże. Tymczasem, dzięki za sugestię!

zbigniewmotyka, 2025-03-15:
Cześć, integracja dostaje informacje z falownika odnośnie dziennej ilości energii jaka została wprowadzona do baterii i jaka z tej baterii została pobrana. Home Assistant w "energia" przyjmuje pierwszy odczyt jako stan początkowy i od tego stanu liczy w każdej godzinie różnicę. Na tej podstawie obliczył że zostało pobrane w tym dniu 0.3kwh energii z baterii.

Jutro pomiary w HA będą pokrywać się z pomiarami w aplikacji solplanet.

Jurny123, 2025-03-16:
> Cześć, integracja dostaje informacje z falownika odnośnie dziennej ilości energii jaka została wprowadzona do baterii i jaka z tej baterii została pobrana. Home Assistant w "energia" przyjmuje [...]

okej, wygląda na to, że nie do końca zrozumiałem, jak ten schemat "dystrybucja energii działa". on nie pokazuje chwilowych parametrów tylko kumulacyjne. Dzięki!

closed 2025-04-15 by zbigniewmotyka