zbigniewmotyka/home-assistant-solplanet #45: Setting soc-min/max with number.set_value

ato-dev-80, 2025-03-16:
Hi, not sure if this in issue with the integration or my understanding is incorrect.
From my perspective it should be possible to set the NumberEntity of soc_min from an automation:

```yaml
automation:  
  - alias: Switch on PV Excess Charging 1P
    description: ""
    triggers:
      - trigger: state
        entity_id: sensor.pv_excess_charging_button
        to: "on"
    condition: []
    actions:
      - action: number.set_value
        target:
          entity_id: number.solplanet_battery_<myinverter_id>_soc_min
        data:
          value: 100
    mode: single
```

However, this does not seem to work (do not get any errors though). Setting the value manually through the predefined input-fields seems to work properly. 

Any idea what could be wrong here?

ffavero, 2025-03-17:
have you tried with quoting the "100"?

andpaw88, 2025-03-17:
> Hi, not sure if this is an issue with the integration or my understanding is incorrect. From my perspective it should be possible to set the NumberEntity of soc_min from an automation:
> 
> automation:  
>   - alias: Switch on PV Excess Charging 1P
>     description: ""
>     triggers:
>       - trigger: state
>         entity_id: sensor.pv_excess_charging_button
>         to: "on"
>     condition: []
>     actions:
>       - action: number.set_value
>         target:
>           entity_id: number.solplanet_battery_<myinverter_id>_soc_min
>         data:
>           value: 100
>     mode: single
> However, this does not seem to work (do not get any errors though). Setting the value manually through the predefined input-fields seems to work properly.
> 
> Any idea what could be wrong here?

Why do you want to set soc_min as 100?

ato-dev-80, 2025-03-17:
As usual the problem was before the screen. I used "sensor.pv_excess_charging_button" instead of "input_boolean.sensor.pv_excess_charging_button". Strange that I did not get any error message from home assistant.
After fixing my script, it now works properly.

@ffavero Thanks anyway for your help.

@andpaw88 I set soc_min to 100 to disable battery charging, which I can use when charging my electric car.

closed 2025-04-15 by zbigniewmotyka