zbigniewmotyka/home-assistant-solplanet #52: Cannot_connect - wersja v0.7.2-beta1

Parley33, 2025-05-21:
Próbowałem połączyć się na paru wersjach i na każdej mam komunikat "Cannot_Connect"

model falownika ASW TLC5000

zbigniewmotyka, 2025-10-29:
Czy możesz sprawdzić, czy wersja v0.8.0-beta1 umożliwia podłączenie się do falownika? Jest tam dodane wsparcie dla API v1.

Parley33, 2025-10-31:
takie 2 błędy w logach:
[image]

[image]

Parley33, 2025-10-31:
żadne z nich nie odpowiada. Z tego co sprawdziłem to nie ma na tym adresie żadnych otwartych portów mimo, że normalnie da się do niego połączyć przez ten adres, a w samym falowniku nie znalazłem żadnych takich opcji czy czegoś podobnego aby otworzyć porty